SunDevilRocketry/Flight-Computer-Firmware #17: Preset Flash full test

niekky, 2025-01-28:


ETSells, 2025-01-28:
Issue identified: Presets have not been removed from the main sensor data in flash extract. Assigning to myself, will attempt to have fix ASAP.

ETSells, 2025-01-28:
(Other relevant branch is [ES/Flash_Extract_Preset_Support](https://github.com/SunDevilRocketry/sdec/pull/7) but that's linked in another repo and I can't attach it straight up).

ETSells, 2025-01-28:
Presets now removed from sensor frame extraction. Issue persists.

ETSells, 2025-02-06:
Test results (2/6):
--erased flash--
presets: 255	255	0.0	0.0	0.0	0.0	0.0	0.0	0.0	0.0	255	255	255	255	
read-preset: fails
sensor data: erased_flash_sensor_data.txt (attached)
--rerun--
presets: 1	4	0.0	3.388036012649536	9.319376945495605	0.0	0.0	0.0	95350.6796875	38.036258697509766	42	48	0	0	
read-preset: Acceleration: [0.0, 3.388036012649536, 9.319376945495605]
Gyro: [0.0, 0.0, 0.0]
Baro: [[95350.6796875](tel:953506796875), 38.036258697509766]
Servo: [42, 48, 0, 0]
sensor data: re_ran_sensor_data.txt (attached)

[re_ran_sensor_data.txt](https://github.com/user-attachments/files/18697887/re_ran_sensor_data.txt)
[erased_flash_sensor_data.txt](https://github.com/user-attachments/files/18697888/erased_flash_sensor_data.txt)

final test:
configured sensors, then ran pid-run and flash extract. results on line 1 of attached.
re-ran pid-run and flash extract. results on line 2 of attached.

[preset_values.txt](https://github.com/user-attachments/files/18697897/preset_values.txt)

ETSells, 2025-02-08:
The listed development branch has been merged. Are we good to close this out?

closed 2025-02-11 by niekky