SunDevilRocketry/Flight-Computer-Firmware #42: Test launch detection in Canard

ETSells, 2025-03-18:
Includes potentially calibrating the sensitivity of launch detection based on AMRAAM and mini.

ETSells, 2025-03-19:
https://github.com/SunDevilRocketry/test/pull/3

New mod test created for this file. This will satisfy low-level testing requirements if #40 does not get completed. If it does get finished, that test can be easily updated to satisfy low-level testing requirements, and then we can do the hardware integration test without needing to be as rigorous about low-level requirements.

ETSells, 2025-03-27:
This issue will track the integration testing. Unit testing is first, which is in progress by David.

ETSells, 2025-04-04:
Integration testing:

Ideally, we want to get both conditions checked. This requires the use of a pressure chamber. I am unable to procure the necessary equipment, so we may postpone part of this to next week as part of system testing. However, we can get accel verified far easier. Here's the general idea:

The test vectors from David's unit tests can be modified to ensure our precise conditions trigger the launch detection. The integration tests just need to a) help determine a suitable threshold for accel and baro (ideally accel being more sensitive than baro) and b) ensure that the vehicle correctly enters PID run upon activation of launch detect. The procedure for this can be open ended, but I recommend trying out (and logging) different manually achievable thresholds, then coordinating with me (@ETSells) and nick (@niekky) about what our final thresholds should be.

ETSells, 2025-04-04:
(unassigning @dmortloff because this was never really his issue to begin with, he did the unit testing)

ETSells, 2025-04-07:
Procedure (accel): 

1. Set threshold = 15.
2. Set num frames = 5.
3. Run the test.
4. Observe from data dump that the threshold was exceeded for 6 frames.

1. Set threshold = 20.
2. Set num frames = 3.
3. Run the test.
4. Observe from data dump that the threshold was exceeded for 4 frames.

ETSells, 2025-04-07:
Accel success!

[flight_comp_rev2_sensor_data.txt](https://github.com/user-attachments/files/19639859/flight_comp_rev2_sensor_data.txt)

Observe lines 9-16 for positive result.

[flight_comp_rev2_sensor_data.txt](https://github.com/user-attachments/files/19639885/flight_comp_rev2_sensor_data.txt)

Observe lines 20-23 for positive result.

Baro has been compiled out due to our inability to access a vacuum chamber.

dmortloff, 2025-04-09:
Pulled out trust calculator to check accel squared values and confirm with acc detect threshold. RIP barometer detection, will fly someday. Launch Detect tests look good to me.

ETSells, 2025-04-09:
Closing as complete on review 🥳

closed 2025-04-09 by ETSells