SunDevilRocketry/Flight-Computer-Firmware #43: Test fin/imu calibration in Canard

ETSells, 2025-03-18:


ETSells, 2025-03-18:
A change needs to be implemented to fin calibration that blocks this issue. PRs are linked here.
https://github.com/SunDevilRocketry/sdec/pull/21
https://github.com/SunDevilRocketry/Flight-Computer-Firmware/pull/44

ETSells, 2025-03-20:
Manual testing performed 3/19/25. I'm going to get started on it using sdec#20 and FCF/Canard-Development. Acceptance of results should wait until the SDEC branch is merged.

GOALS:
- Validate successful saving of presets
- Test boundary conditions on both sides

DEFERRED TO #22:
- Servo actuation/accuracy

PROCEDURE:
1. Reset servo angles to 45.
2. Tune the four servos to an identifiable angle (e.g. {+1, +2, +3, +4}). Save and then read the preset. Verify that the new result matches your inputs (e.g. {46,47,48,49}).
3. Test the lower boundary condition on any of the four servos.
4. Test the upper boundary condition on any of the four servos.
5. Verify that the status LED changes as expected (e.g. green = idle, white = tuning mode).

ETSells, 2025-03-20:
First run:

1. 

[image]

2.

[image]

3.

Step 1 (at the lower boundary): 

[image]

Step 2 (pass below lower boundary):

[image]

Bug identified, ending test.

Cause: uint8_t is explicitly unsigned. If you try to subtract 1 from 0, it wraps back up to the maximum value. This is then captured by the upper boundary, and the uint8_t is set back to the upper bound (180).

Resolution: The upper bound should only snap values if they are within 38 of the upper bound. The lower bound should take everything above that. Implementing fix.

ETSells, 2025-03-20:
Second run:

1. 

[image]

2.

[image]

3.

Step 1 (at the lower boundary):

[image]

Step 2 (pass below lower boundary):

[image]

(Verified bug resolution)

4.

Step 1 (at the upper boundary):

[image]

Step 2 (pass above upper boundary):

[image]

5.

Fin Setup:
[image]
Idle:
[image]

6.

SDEC:

[image]

GDB:

[image]

I'm happy with these results! I'll get someone else to sign off before we close this @SunDevilRocketry/firmware-reviewers

ETSells, 2025-03-25:
Just a note: The branch has been merged, for this one I just want to wait on someone to agree that my procedures tested the calibration adequately and the results look good lol.

niekky, 2025-03-26:
Procedures look good to me. Require testing with actual servos before closing

closed 2025-03-26 by niekky

ETSells, 2025-03-26:
I figure you changed your mind considering you closed this out, but my thoughts were like so:

This is a low level test intended just to test that the servo offsets are modified as expected. The high level testing (#22 i believe) will include a test of the offsets with the servos connected as you seem to indicate here.

niekky, 2025-03-26:
Sounds good!