SunDevilRocketry/Flight-Computer-Firmware #45: Test IMU Calibration in Canard

ETSells, 2025-03-18:


ETSells, 2025-03-20:
Simple test here. We want to make sure that the ImuCalibrationSWCON function reduces the inaccuracies in the IMU readings to just noise.

PROCEDURE:

1. Test orientation with FC flat. Verify no significant drift.

2. Test orientation with FC on its side. Verify no significant drift.

3. Test orientation with FC standing up (roughly launch orientation). Verify no significant drift.

For each step, post both the IMU offset and a few frames of sensor data.

ETSells, 2025-03-20:
Test results: 3/20

1.

[image]
(not quite flat but it's close enough)

2.

[image]

3.

[image]

Hotfixes required are in ES/IMU-Cali-Hotfixes.
- Save IMU and baro presets after auto-calibration.
- Write presets to flash
- Remove absolute value call in calibration.

ETSells, 2025-03-20:
NOTE: The IMU dimensions are as follows:

[image]

niekky, 2025-03-26:
Results look very good! I added some minor feedback to #48. Either way, It should be good to merge in

closed 2025-03-27 by ETSells