SunDevilRocketry/Flight-Computer-Firmware #46: Verify Canard Flash Integrity

ETSells, 2025-03-18:
BUG: There's an issue with flash extract that is unique to Canard. If we can debug this and identify a fix, it would make this a lot easier. Note that this should not prevent us from launching, it just makes extracting flight data post-launch very inconvenient.

ETSells, 2025-03-27:
This is going to be a simple one since we've extensively tested this functionality as we've worked on it. Goals:
- Make sure SDEC still allows us to use flash extract
- Get accurate preset data after calibrating servos and doing auto IMU/baro calibration
- Get accurate sensor data during PID run
- Make sure David's parser script converts to CSV properly
- Measure runtime before timeout

We can test these in two runs.

Run 1:

1. Clear flash
2. Power cycle
3. Calibrate servos to a set value ({46,47,48,49}?)
4. Short the switch terminals, and let it auto calibrate
5. Let it run for a few seconds, and then power cycle
6. Flash extract.
7. Parse the outputted sensor data, and post the output here. Post the preset data.
8. Check the first frame, a frame in the middle, and the last valid frame in flash. Make sure every value is reasonable.

Run 2:

1. Make a quick source code change to lower the launch detect threshold to near-zero.
2. Clear flash
3. Power cycle
4. Calibrate servos to a set value ({44,43,42,41}?)
5. Short the switch terminals and let it auto calibrate
6. Shake the flight computer so launch is detected.
7. Let it run until timeout, then power cycle
8. Flash extract
9. Parse the outputted sensor data, and post the output here. Post the preset data.
10. Check the first frame, a frame in the middle, and the last valid frame in flash. Make sure every value is reasonable.

ETSells, 2025-03-27:
First attempt at run 1 failed. Implementing a hotfix for the flash extract bug and we'll see what happens from there.

ETSells, 2025-03-27:
Run 1 data

[active_roll_preset_data.txt](https://github.com/user-attachments/files/19477764/active_roll_preset_data.txt)
[parsedCSV.csv](https://github.com/user-attachments/files/19477765/parsedCSV.csv)
[flight_comp_rev2_sensor_data.txt](https://github.com/user-attachments/files/19477766/flight_comp_rev2_sensor_data.txt)

I'm happy with it but obviously someone else should review this data. The first frame has weird state estimations and looks like power came off at the end of the last frame, but otherwise it looks good!!

ETSells, 2025-03-27:
Stress test failed at the end due to improper handling of exceeding the flash buffer. Will re-run tomorrow.

ETSells, 2025-03-27:
The bug where we're exceeding the flash buffer persists. Will continue investigating

ETSells, 2025-03-27:
Run 2:

9.63 minutes of runtime

[active_roll_preset_data.txt](https://github.com/user-attachments/files/19494467/active_roll_preset_data.txt)
[flight_comp_rev2_sensor_data.txt](https://github.com/user-attachments/files/19494465/flight_comp_rev2_sensor_data.txt)
[parsedCSV.csv](https://github.com/user-attachments/files/19494466/parsedCSV.csv)

ETSells, 2025-03-27:
I'm happy with this. The only other change I can think for flash is to emulate data logger and log 50 frames or so before clearing and going again (until launch detect).

ETSells, 2025-04-02:
Let's hold the review for now. I'd like to re-run the stress tests when #52 is finished.

ETSells, 2025-04-03:
Moving up to Week 3 due to the need to re-run these tests.

ETSells, 2025-04-03:
Re-run results:

Test #1:

[active_roll_preset_data.txt](https://github.com/user-attachments/files/19595420/active_roll_preset_data.txt)
[flight_comp_rev2_sensor_data.txt](https://github.com/user-attachments/files/19595419/flight_comp_rev2_sensor_data.txt)

Test #2:

[active_roll_preset_data.txt](https://github.com/user-attachments/files/19595620/active_roll_preset_data.txt)
[flight_comp_rev2_sensor_data.txt](https://github.com/user-attachments/files/19595619/flight_comp_rev2_sensor_data.txt)

ETSells, 2025-04-03:
@niekky @journeyhancock once either of you sign off, we can close this out. blocking ticket has been cleared.

journeyhancock, 2025-04-08:
Doesn't look any different from the output I saw when we were working on adding PID frames. So as far as I can tell it looks good.

ETSells, 2025-04-08:
Sweeet okay *finally* closing this issue out.

closed 2025-04-08 by ETSells