SunDevilRocketry/Flight-Computer-Firmware #47: Bug investigation: Canard Flash Extract

ETSells, 2025-03-20:
Issue: There's a discrepancy in the performance of the "flash extract" terminal command between the Canard application and the Data-Logger application. Data-Logger sees a near 100% success rate while Canard is... far below that. 

Assume that SDEC is not at fault. Investigate the difference between how terminal commands (particularly "flash extract") are handled in both firmware applications. Draft a potential solution. If you have time, implement this fix and test its efficacy.

ETSells, 2025-03-20:
Jumping on for an initial investigation. Will report progress.

ETSells, 2025-03-20:
Notes:
Looking at the main loop's USB receive first. This isn't where a flash extract command is handled, but I'm going to look regardless.

- HAL_DEFAULT_TIMEOUT is 10, whereas in data-logger, cases where this should be are occasionally hardcoded to 100.
- The USB receive in canard/main takes an argument of type STATE_OPCODE* (uint8_t enum) when it's a uint8_t in data-logger/main

Potential cause identified:
- In terminal_exec_cmd, usb_receive waits for a length of time defined by HAL_TIMEOUT. This amount of time may be insufficient. Interestingly, there is no error handling for a case where the FC times out and the usb_recieve function would return something that isn't USB_OK. For the sake of debugging, I will be setting the LED to red if there is a timeout using HAL_TIMEOUT. If this succeeds, I intend to define a new constant "USB_TIMEOUT" in main.h with the same value used in data-logger. Flash extract is not as time sensitive as data logging or the PID loop, so we can afford this additional overhead for the sake of reliability.

Will be testing in the shop this afternoon and will report back.

ETSells, 2025-03-20:
Bit of a semantic difference but we can't actually call it USB_TIMEOUT, oops lol

ETSells, 2025-03-20:
Found where the error is being thrown.

main.c:540
case FLASH_OP:
        {
        /* Recieve flash subcommand over USB */
        usb_status = usb_receive( &subcommand         , 
                                  sizeof( subcommand ),
                                  100 );

        /* Execute subcommand */
        if ( usb_status == USB_OK )
            {
            flash_status = flash_cmd_execute( subcommand,
                                                pflash_handle );
            }
        else
            {
            /* Indicate a USB error without stopping the FC */
			led_set_color( LED_RED );
			HAL_Delay(1000);
			led_set_color( LED_PURPLE );
			HAL_Delay(1000);
            }

Where the LED goes red to purple is where the error arises. Looks like it's not one issue, but two. Changing the timeout helps a little bit, but we still get a usb_status error with a timeout of 100.

ETSells, 2025-03-20:
[image]

Can't get past the first block when it fails now. The timeout does seem to have prevented fails mid-extraction. I also seem to have noticed that it's getting fairly consistent on the *FIRST* extract, but it's subsequent flash extracts that throw an error.

ETSells, 2025-03-20:
I'm unable to replicate a lot of my previous issues, so I'm going to leave this open. It's annoying, but canard is still usable despite this bug. Moving to the backlog for now, if anyone has time to tackle it, you're more than welcome to!

ETSells, 2025-03-25:
main.c:341 POTENTIAL FIX: Erase flash and preserve presets (exists in data-logger)

ETSells, 2025-03-27:
Good news: This issue has been resolved with a hotfix in ES/Canard-Hotfixes. This issue will be closed when that branch is merged.

niekky, 2025-04-03:
Looks good to me. But note that one disadvantage of this is that it will erase a section of a data due to the flash hardware characteristic which can lead to data loss. But i'm fine with this solution as we'll have a dedicated flash for the next revision.

ETSells, 2025-04-03:
@niekky It will erase what section of data? My understanding of this is that it will erase everything on the flash chip (intentional behavior) and then write back the presets. There will be a loss of the previous flight's data, yes, but that is the design.

ETSells, 2025-04-03:
Never mind, I'm wrong. I need to prune my old branches when we're done here, I'm confusing myself lol. This functionality is present on the Canard Hotfixes branch.

closed 2025-04-03 by ETSells