SunDevilRocketry/Flight-Computer-Firmware #68: Active Control: Refactor existing roll control implementation

ETSells, 2025-05-31:
## Description
As is, this is fairly difficult to read. I would like for the split between the three relevant functions (flight.c:pid_loop, flight.c:pid_control, and flight.c:v_pid_function) to be refactored and reorganized to promote readability. There are a few parts of this

1. Rename the functions to be more descriptive.
2. Update the descriptions to more accurately describe what each function does.
3. Move around different parts of each function based on what each of them do. 
4. (Optional) Identify code that does not need to be repeated and move it to a setup function to be called before we go into PID.
5. (Optional) Change any helper functions outside of the PID loop and setup functions to static (if you do this one, make sure the CC of each helper function remains very low! Ideally it'll be no more than 3.). This is a good convention to follow in C for production code, but it'll make unit testing harder. Beware!!

## Potential Assignees
This will go to An as her first task. It's a good way to get used to the feature you'll be spending a lot of time with.

Sameer-Pal, 2025-06-01:
HI @ETSells ,
Interested in working.
Please, Clarify which file you referring for **refractoring.**

ETSells, 2025-06-15:
Opening for iter 3: good first issue for new members (low difficulty but maybe a bit too open ended, C task)

ETSells, 2025-07-01:
Pushed back to 4th sprint.

ETSells, 2025-08-25:
No functional changes planned for current release. Expect push back to the next minor release or beyond. Assignee has left firmware team, unassigning.