SunDevilRocketry/Flight-Computer-Firmware #71: Write a function to determine if altitude is decreasing (apogee detection).

ETSells, 2025-06-15:
See epic for more details.

## Description

Using launch_detect.c as a reference, write a function that uses barometric pressure altitude to determine if the vehicle is descending.

### Constraints

1: Ensure that the function runs quickly. This is a subjective measure, but basically avoid invoking HAL_Delay() or using loops.
2: Track the number of times the threshold has been met in a static variable. This will allow us to set a required number of loops that we've been descending to make apogee detection less susceptible to noise. There should be a config option for this.

ETSells, 2025-06-15:
Opening for iter 3: good first issue for new members.

ETSells, 2025-07-01:
Pushing back to sprint 4.

ETSells, 2025-07-16:
@VedantVashi Has there been any progress on this? Deadline for the whole epic is the end of the current sprint.

VedantVashi, 2025-07-16:
Yes just finished this

closed 2025-08-16 by ETSells