SunDevilRocketry/Flight-Computer-Firmware #76: Add the ability to send a camera signal from parachute terminals

ETSells, 2025-06-30:
## Description

Customer Request. AHPR has asked if we can send a signal from the flight computer to trigger their cameras. I suggested that we repurpose the parachute terminals for programs that don't want dual deploy.

## Suggested Implementation

The feature itself will be simple. The difficulty from this task will be setting it up to be sufficiently configurable. We need the use of the terminals to be dependent on uploaded configs. There are a few ways you could do this, but at least two different configurations should be possible:
1- Use both terminals for parachutes.
2- Use both terminals for cameras.

It could also be desirable to include an option where the main is used for parachutes and the drogue is used for cameras. This will be up to you to figure out implementation.

If needed, I'll give you a second sprint for this feature since messing with the config struct can get involved. I'll try to give you a guide to messing with configs soon.

ETSells, 2025-07-01:
[image]

Here are the system requirements

niekky, 2025-07-16:
Marzuk said he doesn't need anything from the FC because they have already E-switch. Will be circled back when have telemetry

NArmistead, 2025-07-16:
Closing this but will leave the NA/camera-control branch up for when we come back to it

ETSells, 2025-07-16:
Reopening

niekky, 2025-07-16:
That sounds good to me. But if requirements change and scope creep happens, I suggest closing this as won't do and create a new ticket with better estimate for it.

closed 2025-08-03 by NArmistead