SunDevilRocketry/Flight-Computer-Firmware #123: Test: Build electrical device to simulate ejection charge

ETSells, 2025-09-02:
## Description

In order to simulate the use of ejection charges, we need a hardware peripheral that can be connected to the flight computer's parachute terminals. Ideally, this will have two functions.

1. Complete the circuit between terminals to verify continuity.
2. Indicate in some way when the voltage is HI.

## Notes

The terminals will assert HI for 1250ms (1.25s) on deployment. Auditory methods are preferred for indication in case this interval is shortened so this can be reused.

There are two terminals that can be ignited, so we would prefer if two copies are made.

The firmware checks for continuity after ignition. If possible, we would prefer that the circuit be closed in some way after the deployment.

ETSells, 2025-09-02:
@dmortloff Bringing this to your attention to be prioritized and opened at your earliest convenience. Note that it's blocking a priority 1 task for us.

dmortloff, 2025-09-02:
If we have any Piezo buzzers I can whip together a morally questionable solution, GPIO pins should output about 15mA nicely and the buzzer only needs 1mA at 5V, so if we have any more i can solder a OSHA violation for you

ETSells, 2025-09-03:
Morally questionable? OSHA violation? Sign me up!