SunDevilRocketry/Flight-Computer-Firmware #137: Preset: Fix checksum verification

ETSells, 2025-09-10:
## Description

One of the intended launch features with APPA was the ability to send data with a checksum and verify it on the other end to ensure data integrity. This was implemented ~70% of the way, but we were unable to get it working fully. This will likely take a few steps:

1: Implement the CRC32c algorithm in C in mod/common/common.c (replacing the crc32 function).
2: Verify that on four bytes of data or so, the output of the python crc32c library is the same as this function.
3: Pair program with @ETSells to get checksum verification working in app/flight/appa/rev2/prelaunch.c.

Expect a sprint or so to do the first and second tasks, then another to do the integration.

ETSells, 2025-09-17:
Opening, medium priority, high difficulty.

favillat, 2025-09-18:
I found the reason our checksum didn't match the python library was simply because they were using two different polys to generate the CRC; we were using 0xEDB88320 while the CRC32C library was using 0x82F63B78.

ETSells, 2025-09-18:
Good find! If you test both functions as scratch programs, do they have the same output?

favillat, 2025-09-18:
yur, I didn't like extensively test it (I'm not super sure how to) but i tried it with a few string inputs and they matched up! 

Here's just one example:
[image]

ETSells, 2025-09-18:
ehhh that looks good lowkey. what program is running those tests?

favillat, 2025-09-18:
I created a new vscode project, copied over the the function as seen in common and created a simple script that takes in data as a string and feeds it into the crc32 function and prints the result. Then i pretty much just brute force tested different inputs on each file lol. maybe not the best way of doing it but it was simple and quick

ETSells, 2025-09-20:
[image]

just found out we have a hardware unit for this lol. our use of the crc isn't performance critical so we'll just use this, but i think it's funny.

That testing procedure sounds good! I did the same when I was working on the gps parser lol, it's a good starting point if we want unit tests on this function in the future (which is likely).